BoiseState-AdaptLab/adapt-lidar-tools #246: Valgrind: "Dwarf2 reader: badly formed extended line op encountered" messages flooding output

SpencerFleming, 2019-07-24:
This has been happening for a while and isn't a critical issue but I think it should be looked into. When Valgrind is run on our code, it will produce hundreds of lines that say "dwarf2 reader: badly formed extended line op encountered". It makes reading the output require a lot of scrolling. Supposedly this means that either Valgrind is reading the code wrong, or we're not producing debug information correctly.

This makes me wonder if we even have debugging flags in our makefile right now.

jaredwhite547, 2020-04-23:
Can't reproduce, and haven't noticed before. Closing.

closed 2020-04-23 by jaredwhite547

dufferzafar, 2022-07-03:
I'm seeing the same issue. when I ran `valgrind --tool callgrind` on our app, it spew out lots of `==2381658== Warning: DWARF2 reader: Badly formed extended line op encountered` errors

dufferzafar, 2022-10-04:
@jaredwhite547 Is there any more debugging information I could provide that would help with this?