BoiseState-AdaptLab/adapt-lidar-tools #249: Peak triggering amplitude

rvishnkr, 2019-07-25:
A Peaks triggering amplitude is the first amplitude in the waveform that is greater than the noise level. 
It is currently however being calculated as `peak->triggering_amp = noise_level + 1;`

Aaron3154518, 2019-07-26:
A noise level of 6 approximately doubles the average number of peaks per waveform. This increased the processing time significantly (19sec to 118sec)

Aaron3154518, 2019-08-04:
The calculation for triggering amplitude is now as follows:
With the equation for the peak with solve for the time at which amplitude = noise_level.
We then round up to the nearest integer. This gives us the first integer time bin at which our gaussian is above the noise level, which is the triggering location
Finally we solve for the amplitude (triggering amplitude) at that time value (triggering location)

closed 2019-08-07 by cathieO