BoiseState-AdaptLab/adapt-lidar-tools #257: Activation locations can be on the edge of the bounding box

rvishnkr, 2019-08-01:
Ahmad had an issue where his activation location was on the edge of the bounding box - 

`x activation: 241122 not in range: 238097 - 241122`
The coordinates of his file are:
```
Min x: 238096.690
Max x: 241122.001
Min y: 4322116.505
Max y: 4322473.613
```

After talking to Nayani, We have decided to include all activation locations 1 meter greater than the bounding box extents.

Aaron3154518, 2019-08-07:
I talked with Nayani and confirmed how the LiDAR modules collects data. I also added code to extract the xyz_last values from the pls/wvs files set. These are the xyz locations of the last data point in each waveform. I will compare the min/max xyz values of the waveforms to that of the bounding box and post results here. This will help us conform the bounding box to our data so that all activation locations are in the bounding box.

rvishnkr, 2019-08-07:
For the time being, we are updating the x and y activations such that they can be located on the bounding box `*_activation > bb_*_max+1`. Removing z_activation check.

closed 2020-04-30 by jaredwhite547