BoiseState-AdaptLab/adapt-lidar-tools #261: Driver for Gaussian Fittting from text file

cathieO, 2019-08-07:
this will be a diagnostic tool to help us figure out how to fit the small/large peak forms

cathieO, 2019-08-14:
1. read waveforms from a file
File format: 
v1 v2 v3 v4...
0 1 2... 61 62

2. for each waveform we will fit it, with diagnostics on.
0: original wave
Guess parameters
1: guess after 1 iteration
New parameters: error estimates
2: guess after 2 iterations...

SpencerFleming, 2019-08-15:
Shortly after I began modeling this epic in a UML diagram, I realized that how I insert the diagnostic features into our Gaussian fitter functions was going to decide a lot of my design choices, and thus, it was a blocker to a top-down approach to designing the feature.

I'll make coming up with options for this into a task.

closed 2019-10-23 by cathieO