BoiseState-AdaptLab/adapt-lidar-tools #275: Implement TxtWaveReader

SpencerFleming, 2019-08-18:
This class can read our own text-based format for communicating raw waveform data, and will be used as a debugging tool to test specific kinds of input data.

The format is defined in the Epic this issue is attached to.

The implementation will generate PulseWave objects as they are requested. The class will throw an exception if a PulseWave object is requested when there isn't more data left in the file, but also, there will be a function to see if there's more data left.

##Acceptance Criteria
- [ ] Files open and close correctly
- [ ] Outputted data reflects data in text files
- [ ] There is appropriate unit test coverage and the unit tests pass

SpencerFleming, 2019-08-19:
6 hrs on this, it's in the testing phase now.

SpencerFleming, 2019-08-21:
6 more hours, all test cases are running and passing. Any bugs I found along the way are gone now.

cathieO, 2019-08-28:
don't use PulseWaves -- rather use a vector of values.

SpencerFleming, 2019-08-31:
TxtWaveReader no longer uses PulseWaves to store the data. There are now two object variables that store the most recently-read idx and wave vectors, respectively.
Other changes:
- next_wave, the function originally responsible for returning PulseWave objects with txtfile data, now simply overwrites the object variable vectors with its txtfile data
- next_wave also returns a bool specifying whether or not there are more lines to read, so it can be used in control statements.
- some print statements I used for debugging, specifically the ones I thought may prove useful down in the future, have been promoted to logger messages. The others were deleted entirely.
All unit tests are passing again, so I believe this can be moved to done once again.

closed 2019-10-23 by cathieO