BoiseState-AdaptLab/adapt-lidar-tools #291: exit( ) called by FlightLineData::getNextPulse

SpencerFleming, 2019-09-01:
exit is called three times by this function:
- When there is no next pulse to be read
- When the type of the first sampling isn't outgoing
- When the type of the second sampling isn't returning

My proposed solution to this is to simply return an empty PulseData object that will resolve to "NO_DATA" when fitted, and use spdlog to report a critical error. There is a good chance that the rest of the data is salvageable if this happens.

SpencerFleming, 2019-09-02:
I changed everything like described above and all unit tests still pass.

7thfleet, 2019-10-16:
Need to return a value that can indicate an error.

jaredwhite547, 2019-10-17:
Need to return a value that can indicate an error.

closed 2020-04-30 by jaredwhite547