BoiseState-AdaptLab/adapt-lidar-tools #297: Turn min/max peak amplitude in GaussianFitter into instance variables

SpencerFleming, 2019-09-02:
In GaussainFitter::find_peaks:
```
if (peak->amp >= 2*max || peak->amp < 10 ||-
    peak->triggering_location > n || peak->triggering_location < 0){
    delete(peak);
    results->erase(iter--);
    } else {
    // set the peak position in the wave
    // this will be wrong if a previous peak was removed for any
    // reason
    peak->position_in_wave = i+1;
 }
```
This throws away peaks if they have amplitudes greater than 2 times the max, or less than 10. This relies on another variable, so I'll leave this in QA and discuss it with others before making a plan.

SpencerFleming, 2019-09-03:
- '2' will be replaced by a reference to the new instance variable amp_upper_bound
- '2' will be the default value, assigned by the constructor
- '10' will be replaced by a reference to the new instance variable amp_lower_bound
- '10' will be the default value, assigned by the constructor

SpencerFleming, 2019-09-03:
Seems to be working just fine with the new changes.

SpencerFleming, 2019-10-09:
We want to think more about how to convert this into an instance variable.

SpencerFleming, 2019-10-13:
This is the current state of the code in question.
```
 if (peak->amp >= amp_upper_bound*max || peak->amp < amp_lower_bound
         || peak->triggering_location > n
         || peak->triggering_location <0) {
     delete(peak);
     results->erase(iter--);
 } else{
    //set the peak position in the wave
    //this will be wrong if a previous peak was removed for any
    //reason
    peak->position_in_wave = i+1;
 }
```
Where amp_upper_bound's default is 2, and amp_lower_bound's default is 10.

SpencerFleming, 2019-10-16:
2 should be named something like max_amp_multiplier, default to 2, be a float, and be added to the advanced options

cathieO, 2019-10-23:
This needs to be added as a command line argument.

SpencerFleming, 2019-11-06:
Make sure to add help option

jaredwhite547, 2020-04-23:
Min peak amplitude is in template refactoring. Need to check if we still want a max amplitude option.

closed 2020-04-30 by jaredwhite547