BoiseState-AdaptLab/adapt-lidar-tools #299: Turn default peak width in GaussianFitter::guess_peaks into an instance variable

SpencerFleming, 2019-09-02:
In GaussianFitter::guess_peaks:
```
754         if(guess<0){
755             guess = 4;
756         }
```
```
762          if(guess > 20){guess = 10;}
763             peaks_found++;
```
The 4, 20 and 10 can all be converted into instance variables.

SpencerFleming, 2019-09-03:
Finding good names for these is a bit difficult;
for now these are what I'm going with:
guess_lt0_default = 4;
guess_upper_lim = 20;
guess_gt_upper_lim_default = 10;

'guess' is the variable being modified here, so anyone familiar with the function these refer to will immediately know what they are. Their names explicitly state their purpose: if guess is less than zero, it defaults to the value of guess_lt0_default; a guess's upper limit is guess_upper_lim; etc.

SpencerFleming, 2019-09-03:
They have been implemented as instance variables now. Default values are assigned by the constructor using macros kept in the header file (a sensible place to put compile time parameters)
My rational for the macros is that:
1) They are easy to understand
2) They are fast
3) They will be easy to replace later when we add ways to change these values at run-time

Aaron3154518, 2019-10-09:
We need to fix the magic numbers on line 719 that determine the max FWHM value and its default.

SpencerFleming, 2019-10-09:
I should change the name of guess_lt0_default; the name doesn't make sense and I'm the one who wrote it.

closed 2019-10-16 by jaredwhite547