BoiseState-AdaptLab/adapt-lidar-tools #336: Check messages of logger

FloCiaglia, 2019-10-09:
Run the logging over and over in different cases and make sure the messages make sense where they are at.

cathieO, 2019-10-23:
Make sure that when running using info --- no extra error messages show up

SpencerFleming, 2019-10-23:
Write down examples to show how my implementation of the messages looks.

SpencerFleming, 2019-12-18:
Changes I have made:

LidarDriver::produce_product - Messages that exist to show us where in runtime the program is have been moved to debug to match other messages of that type in other files.

pls_to_csv.cpp (in main) - When program is able to open input file successfully, an info message would let you know. Now it is a debug message.

TxtWaveReader.cpp - Some debug messages contained information that informed the user that a line was read that did not follow the correct format. These are error messages now, because if the file is formatted incorrectly, the user needs to know.

SpencerFleming, 2019-12-18:
Second wave (based on testing rather than skimming source files):
Was getting:
>[  info  ] Processing etc/140823_183115_1_clipped_test.pls
>[ error  ] Values per product: 9
>[  info  ] Finding peaks with gaussian fitting

The error message is in LidarDriver and should be a trace message. This is now fixed.

Something to consider:
The message "Guess for peak width less than zero, reverting do default" floods logger with messages. This is from Guess Peaks, in GuassianFitter. We wanted situations where we have to revert to a default to be warning messages, so this is correct, but perhaps we can modify the logic leading to the logger message occurring to make it so the warning occurs once and the specific instances are recorded in trace.
To recreate this problem, run geotiff-driver on almost any of the pls files we have with actual data in them.
I'm not changing this in case this is indicative of a deeper problem that needs to be looked into.

The output of running the program seems sensible now, to me at least.

closed 2020-04-30 by jaredwhite547