BoiseState-AdaptLab/adapt-lidar-tools #354: Iterate with Ahmad

cathieO, 2019-10-23:
See if he is happy with the product.

SpencerFleming, 2019-10-23:
1) Ask about workflow
2) Keep in mind it should be used with small files
3) Let him know that fitting is in "a state of flux"

SpencerFleming, 2019-11-01:
To checkout a specific commit, we first clone the repo if we haven't already, and then `git checkout <hash>`, where `<hash>` is the commit in question.

SpencerFleming, 2019-11-01:
I sent a first email touching base and giving Ahmad the general idea of what I'm hoping to do here.

SpencerFleming, 2019-11-06:
Ahmad got back to me, and suggested the following:
- It should support recording the following
  - amplitude
  - pulse width
  - rise time
  - X Y Z coordinates
  - backscatter coefficient
  - peak ID (first, second, last etc)
  - input wave amp
  - pulse width for each peak
- We should have each be a separate file on a NetCDF or HDF format, to help save space on larger files. Both of these are open standards.

SpencerFleming, 2019-11-06:
run CSV driver on a series of files (small, medium, big-ish), choose the options that prints out wave amplitudes (incoming wave)
then do it with just the max peaks for each peak
TIME the runtimes 
Update CSV driver so it is aware of situations when Gaussian fitting is not needed
run csv driver on etc file too, send this to ahmad, ask him if that's what he wants

jaredwhite547, 2019-12-04:
Change from one csv with all attributes, to one csv per attribute. Communicate with Ahmad to figure out what he wants. @FloCiaglia

closed 2020-04-30 by jaredwhite547