BoiseState-AdaptLab/adapt-lidar-tools #357: Converting std::vector to struct

brandonechols97, 2019-11-11:
Improving performance by removing `std::vector` usage in areas that call `push_back`. This change is to use a struct that contains an array with size and max capacity instead. Due to `push_back` allocating memory every time a new element is added.

brandonechols97, 2019-11-15:
[image]
[image]

jaredwhite547, 2019-12-01:
FYI you can prevent std::vector from reallocating by simply calling [reserve](https://en.cppreference.com/w/cpp/container/vector/reserve) with the size you expect. No more allocations will occur unless this size is exceeded.

jaredwhite547, 2020-04-23:
Reverted. New refactoring makes this unneeded.

closed 2020-04-23 by jaredwhite547