BoiseState-AdaptLab/adapt-lidar-tools #359: Incorrect calculation of c!!!!!!!

Aaron3154518, 2019-12-08:
In first differencing, we calculate the FWHM by solving the equation of a Gaussian given our amplitude, amp location, and a data point on the gaussian. However we then calculate c in Gaussian Fitting by dividing the FWHM by 2log(2), which is incorrect. The relationship between c and width is as follows:

w = 2sqrt(2ln(n)) * c given w is the width when y = Amp/n.

Therefore with the FWHM, n = 2 and c = w / 2sqrt(2ln(2)).

However we do not need to calculate these two variable this way. Here is an alternative method:

Given data point (xi, yi), amplitude a, and amp location t,
xi - t is half of the full width (w) at yi
yi can be put in terms of A/n by noting that yi = A * yi/A
This means 1/n = yi/A and n = A/yi and w = 2sqrt(2ln(A/yi)) * c and c = w / 2sqrt(2ln(A/yi))
Now we know that for the FWHM, n = 2 and so FWHM = 2sqrt(2ln(2)) * c
When we substitute c, FWHM = 2sqrt(2ln(2)) * w / 2sqrt(2ln(A/yi)) = w * sqrt(ln(2) / ln(A/yi)) = 
(xi - t) * sqrt(ln(2) / ln(A/yi))

Thus we can use ANY data point to calculate the FWHM as that point gives us yi and xi and we know A and t.
Then in gaussian fitting, we simply note that c = w / 2sqrt(2ln(2)).
This gives us a much more accurate answer for FWHM and thus c as we can use the actual data instead of the closest point to half maximum.

Aaron3154518, 2019-12-10:
Additionally in the formula y = Ae^-1/2((t-b)/c)^2, c is the standard deviation.
If we were to treat each gaussian as a histogram and calculate the standard deviation from that, we could compare that to c to determine if we even need to fit the curve.
(Perhaps only for single peak waveforms)

Aaron3154518, 2019-12-10:
Gaussian Curve info:
https://en.wikipedia.org/wiki/Gaussian_function

jaredwhite547, 2020-02-28:
I think this has been resolved.

closed 2020-04-23 by jaredwhite547